EEVengers/ThunderScope #211: Get Windows Driver Signed to Run Without Test Mode On

AleksaBjelogrlic, 2021-11-29:
I don’t know what this entails, but we should definitely figure it out so our Windows users don’t have to keep test mode on all the time (I’ve learned the hard way that you can’t play most games with it enabled due to anti-cheat).

AleksaBjelogrlic, 2025-02-28:
Need a windows dev membership or something of the sort (essentially just give microsoft money)

NateMeyer, 2025-03-01:
Perhaps look into [Azure Trusted Signing](https://azure.microsoft.com/en-us/products/trusted-signing)? I'm not sure if Microsoft has this working with submitting signed kernel drivers or not? I've seen mixed answers online so far.

AleksaBjelogrlic, 2025-04-19:
Signed up for the Microsoft Hardware Developor Program

AleksaBjelogrlic, 2026-08-31:
Microsoft signed driver is available here: https://github.com/EEVengers/ts_litex_driver_win/releases/tag/v1.0.0 - closing this issue.

closed 2026-08-31 by AleksaBjelogrlic