EEVengers/ThunderScope #212: Add Linux Support

AleksaBjelogrlic, 2021-11-29:
Should be able to IFDEF in the Xilinx XDMA Linux driver, the rest of the code is cross-platform (for glscopeclient that is - the electron UI that we’re switching away from uses Windows named pipes)

profezzorn, 2021-12-02:
A little bit more context would be helpful here.
Do you mean this driver? https://github.com/Xilinx/linux-xlnx/blob/master/drivers/pci/controller/pcie-xdma-pl.c
What #ifdefs are you referring to?
Is there a way to make this work today, or is it only possible on windows right now?

AleksaBjelogrlic, 2021-12-02:
My bad, I should have linked the repo. Here it is: https://github.com/Xilinx/dma_ip_drivers/tree/master/XDMA/linux-kernel
The processing code should work once this driver is in. I think we already have some ifdefs for windows vs. Linux, but essentially we want to load this driver on Linux and the other one on windows - they should have very similar functions, so the changes likely won't extend past PCIe.cpp.

profezzorn, 2021-12-02:
What PCI device ID does the thunderscope use?
(Or, where do I find it?)
I tried looking at the windows driver, but it wasn't obvious.

AleksaBjelogrlic, 2021-12-02:
That's set by the PCIe IP on the FPGA - I'll post the info here later today when I get back from work

profezzorn, 2021-12-03:
The kernel driver seems to be easy to compile, and it should be trivial to add whatever PCI ID we need to it.
However, I'm not sure how to make PCIe.cpp work on linux without hardware to test it on.
Maybe if there is a linux version of the example programs you used to write PCIe.cpp, I can try to integrate?
Most of the example code in the XDMA repository don't seem to have a way to enumerate devices, they just ask for a device file and call it good....

profezzorn, 2021-12-31:
Should we close this now, or is there more to do?

AleksaBjelogrlic, 2021-12-31:
Closed! All that's left for Linux is getting glscopeclient to work with the processing software (scope_link), but that's covered by another issue.

closed 2021-12-31 by AleksaBjelogrlic