EEVengers/ThunderScope #222: thunderscope identification chunk

profezzorn, 2022-01-04:
It would be great if we could have:

1. a magic number to identify without a doubt that we're talking to a thunderscope, not some other xilinx-based device.
2. hardware revision (major/minor)
3. firmware revision (major/minor)
4. serial number

Accessible from the USER handle. This would let us keep the software compatible if/when we make new versions of the hardware and firmware, and it would simplify enumeration and selection of scopes in those cases where is more than one scope present.

profezzorn, 2022-01-05:
One more thing that would be handy to get from the scope somehow: The size of the RAM buffer that we're fetching data from.
It could just be a table in the client based on the hardware major/minor I suppose, but getting it from the scope would be nice I think.

AleksaBjelogrlic, 2025-02-28:
1. PCI VID
2. Pin straps added to Rev 5
3. Already have this in LiteX gateware
4. Should add this if not already in LiteX gateware

NateMeyer, 2025-04-06:
Added EEVengers VID and PID for ThunderScope.

PID for ThunderScope will use `0101`.

Preliminary plan for assigning PIDs:
 [15..12] Category 0, first used for EEVengers devices
 [11.. 8] Class of Device, 1: Oscilloscope, others TBD
 [ 7.. 0] Unique product index, 01 for ThunderScope 

https://github.com/EEVengers/thunderscope_litex/commit/1735eff9ae7453a85be40eb02bebbd6f7980a4e7