EEVengers/ThunderScope #292: Ref Clock In/Out

AleksaBjelogrlic, 2025-03-02:
Starting this issue to centralize technical discussions on this here - moving away from using the discord for those.

I think we output the clock by default, so we need to try the clock input config. Would be great to give the user more control over the clock gen as well, so they could sample synchronously with their DUT at an arbitrary frequency/phase as well as the more typical 10MHz used with other test equipment.

NateMeyer, 2026-01-02:
Added Ref Clock configuration to libtslitex with https://github.com/EEVengers/libtslitex/commit/6d06605e6ca0d8312c480f3f5481d52b3fcf29ab

Tested in https://github.com/EEVengers/ThunderScope/issues/419

AleksaBjelogrlic, 2026-01-02:
Thanks for updating the issue @NateMeyer, closing this now

closed 2026-01-02 by AleksaBjelogrlic