EEVengers/ThunderScope #303: PCIe ESD Protection / Warnings

AleksaBjelogrlic, 2025-05-09:
> NOTE: I had to replace the TPS563203 VUSB Buck after it stopped working during this test. I'm not sure what caused it but I'm suspecting ESD to the EN pin, which is connected directly to 12V. I will open another issue for this. 

 _Originally posted by @AleksaBjelogrlic in [#302](https://github.com/EEVengers/ThunderScope/issues/302#issuecomment-2865141188)_

Investigate this further. Consider adding ESD protection to the 12V input or a divider and protection on the EN pin. I don't want to risk any trouble with PCIe signal integrity on the actual PCIe signals so I will add an ESD warning symbol by the edge connector to warn users to handle with care.

azonenberg, 2025-05-09:
You definitely need a divider. Absolute max on the EN pin in the datasheet is 6V.

[image]

AleksaBjelogrlic, 2025-05-09:
Good catch, will add one in rev 5.1!

AleksaBjelogrlic, 2025-05-09:
Bodged on a 20k-10k divider on Rev 5 Unit 4:

[image]

EN pin measures 3.8V so it should be happy now