EEVengers/ThunderScope #314: Front end noise greater than expected

AleksaBjelogrlic, 2025-05-25:
After trimming the offset on unit 4, I measured the noise at the 8mV range and it's worse than rev 4.1:
 
[image]

Just to rule out any ADC weirdness, I replaced the ADC on unit 3 (which already had an issue with the ADC) with a fresh ADC bought directly from mouser. This is a TS-USB4 unit so I powered it from VUSB using the bench supply directly, the FFT is a bit different as a result:

[image]

I suspect it's once again the relays, there may not be adequate filtering on them this rev. Will try playing around with those next

AleksaBjelogrlic, 2025-05-25:
As expected, disconnecting *only the termination relay* from VUSB on unit 4 resulted in a huge noise improvement:

[image]

AleksaBjelogrlic, 2025-05-25:
Added a ferrite in series with the remaining 30 Ohm resistor on the attenuator relay:

[image]

And here's the result with the input shorted (this is the standard for noise measurement):

[image]

Not much change from the ferrite, which is kinda expected, it's a 120Ohm@100MHz part. Will remove it and try doubling up on the caps instead

AleksaBjelogrlic, 2025-05-25:
Chonky 22uF caps installed

[image]

AleksaBjelogrlic, 2025-05-25:
We may be at the point of diminishing returns here, RMS measurement with shorted input below:

[image]

AleksaBjelogrlic, 2025-05-25:
2nd order low pass, why not?

[image]

AleksaBjelogrlic, 2025-05-25:
Looks like we can't get much better than this from the relays alone:

[image]

I'll do this same mod on the termination relay and see if the noise improvement holds up

AleksaBjelogrlic, 2025-05-25:
Still not quite there:

[image]

sslupsky, 2025-05-25:
Is that the DCDC converter switching frequency?

AleksaBjelogrlic, 2025-05-25:
Yup, for the 12V to 5V3 buck

AleksaBjelogrlic, 2025-05-25:
Nope! Next!

[image]

AleksaBjelogrlic, 2025-05-25:
Doubled up on the output caps 
[image]

AleksaBjelogrlic, 2025-05-25:
Better!

[image]

AleksaBjelogrlic, 2025-05-25:
Nope, **much worse**

AleksaBjelogrlic, 2025-05-25:
Doubled Cff, it was already a bit smaller than recommended

AleksaBjelogrlic, 2025-05-25:
Wow that's a lot of ripple:

[image]

[image]

AleksaBjelogrlic, 2025-05-25:
Cff change seemed to tame it a bit:

[image]

sslupsky, 2025-05-25:
Is R71 installed?

AleksaBjelogrlic, 2025-05-25:
> Is R71 installed?

Yup

AleksaBjelogrlic, 2025-05-25:
This is better:

[image]

sslupsky, 2025-05-25:
> > Is R71 installed?
> 
> Yup

I cannot find an image of the assembled board so I cannot tell if the boost converter is populated?  Could that be interfering?

AleksaBjelogrlic, 2025-05-25:
> > > Is R71 installed?
> > 
> > 
> > Yup
> 
> I cannot find an image of the assembled board so I cannot tell if the boost converter is populated? Could that be interfering?

The boost converter is not populated on the PCIe variant (which is what unit 4 is)

AleksaBjelogrlic, 2025-05-25:
Here's a comparison with channel 1 (that has no mods to the relay filters):

[image]

We do get rid of these spikes but it looks like the majority of the noise is coming from big gulps of power going to the FPGA

AleksaBjelogrlic, 2025-05-25:
My intuition here is there's a load regulation problem with the VBUS regulator. I removed the extra input and output caps, keeping only the increased cff.

[image]

AleksaBjelogrlic, 2025-05-25:
> Wow that's a lot of ripple:
> 
> [image]
> 
> [image]

I think this was a measurement error. I see the same thing when the probe isn't making good contact with the reg

AleksaBjelogrlic, 2025-05-25:
Swapped back to 10uF on the one relay, want to see if that's actually enough since I think this issue is no longer relay based.

[image]

AleksaBjelogrlic, 2025-05-25:
Maybe the 22uF should stay lol:

[image]

[image]

AleksaBjelogrlic, 2025-05-25:
Well I don't think it's actually helping in the "don't couple through the relays" way

AleksaBjelogrlic, 2025-05-25:
Why not, I'll swap the other 22uF's for 10uFs and see

AleksaBjelogrlic, 2025-05-25:
You can def see some switching transients now

[image]

AleksaBjelogrlic, 2025-05-25:
Hows the regulator going to react when every relay has 22uFs in the filter? Let's find out!

AleksaBjelogrlic, 2025-05-26:
Not bad, figured we could use a sort of distributed bulk cap

[image]

AleksaBjelogrlic, 2025-05-26:
Reconnected the buck reg.

Here's with a 220uF alupoly bulk cap:

[image]

and without:

[image]

The noise jumps around a bit, but there is essentially no change here

AleksaBjelogrlic, 2025-05-26:
I've depopulated R71 to feed the front end 5V3 directly from the supply:

[image]

As expected, all the low frequency content (<100kHz) is from the FPGA doing it's thing

AleksaBjelogrlic, 2025-05-26:
Maybe??

[image]

AleksaBjelogrlic, 2025-05-26:
Bodge caps will continue until RMS noise improves

AleksaBjelogrlic, 2025-05-26:
I probed all the FPGA rails and can't see anything on my setup

AleksaBjelogrlic, 2025-05-26:
Added a 47uF on the FPGA 1V8 Rail

AleksaBjelogrlic, 2025-05-26:
No real improvement:

[image]

AleksaBjelogrlic, 2025-05-26:
Extra 47uF on the 2V5 Rail

AleksaBjelogrlic, 2025-05-26:
At this point the noise doesn't go over 70uV RMS very often, so an improvement, but a small one

AleksaBjelogrlic, 2025-05-26:
Mmmh, I think all the extra caps on the FPGA rails did actually do something, look at the FFTs!

[image]

AleksaBjelogrlic, 2025-05-26:
> Maybe??
> 
> [image]

This one...

AleksaBjelogrlic, 2025-05-26:
> No real improvement:
> 
> [image]

Versus this one! Much less junk in the low frequencies

AleksaBjelogrlic, 2025-05-26:
> Mmmh, I think all the extra caps on the FPGA rails did actually do something, look at the FFTs!
> 
> [image]

Anyway, back to this. I think the switching noise is our primary problem again here

AleksaBjelogrlic, 2025-05-26:
Just for completeness, I'll add some extra caps on the MGT rails

AleksaBjelogrlic, 2025-05-26:
Added a 47uF on 1V2_MGT

AleksaBjelogrlic, 2025-05-26:
> > Mmmh, I think all the extra caps on the FPGA rails did actually do something, look at the FFTs!
> > [image]
> 
> Anyway, back to this. I think the switching noise is our primary problem again here
> 
> [image]

Compare and contrast with this

AleksaBjelogrlic, 2025-05-26:
> Removed the 10R resistor to the termination relay and the spikes are gone again... these damn relays
> 
> [image]

compare and contrast with 

[image]

AleksaBjelogrlic, 2025-05-26:
Oh wait the whole time I had the attenuator and termination relays mixed up, it's the **attenuator** relay that's super sensitive to VUSB

AleksaBjelogrlic, 2025-05-26:
Tried feeding the attenuator relay from the 10R resistor of the termination relay:

[image]

AleksaBjelogrlic, 2025-05-26:
Going full send and using a 100pF CFF (max recommended value from the datasheet) seems to fix the instability. Still have the spikes though

[image]

[image]

AleksaBjelogrlic, 2025-05-26:
I've removed power from the termination relay this time, still spikes

[image]

AleksaBjelogrlic, 2025-05-26:
Ok, just to realllly put a bullet in any non-relay theory - with the termination relay unpowered, I've powered the attenuator relay with the bench supply directly:

[image]

AleksaBjelogrlic, 2025-05-26:
> Holy shit it might actually be layout

I've thought this many times, rarely was it actually the case

AleksaBjelogrlic, 2025-05-26:
I can't believe I'm doing this mod again

[image]

AleksaBjelogrlic, 2025-05-26:
## The plot thickens!

With no VUSB going anywhere near the front end now, and the CH4 attenuator relay powered by the bench supply, we still see spikes at the switching frequency!

[image]

AleksaBjelogrlic, 2025-05-26:
> I've depopulated R71 to feed the front end 5V3 directly from the supply:
> 
> [image]
> 
> As expected, all the low frequency content (<100kHz) is from the FPGA doing it's thing

No spikes here, so it must be getting in through the front end rails!

AleksaBjelogrlic, 2025-05-26:
Or maybe there are?? I'll test again since that screenshot doesn't tell the full story

AleksaBjelogrlic, 2025-05-26:
They are in fact gone. Def looks like the front end rails then

AleksaBjelogrlic, 2025-05-26:
Finished that mod from rev 4.1

[image]

[104 more replies]

closed 2025-09-19 by AleksaBjelogrlic