EEVengers/ThunderScope #324: Confusing commits

Vrail99, 2025-06-17:
Hi,

I did a fresh install on a Ubuntu 24.04 machine (BTW: Kernel 6.11 is not compatible with TS, had to manually install kernel 6.8).
Now I get this message, when using the latest libtslitex:

`09:42:10 info: TS.NET.Engine[0] x86/x64 CPU with AVX2.
Unhandled exception: System.EntryPointNotFoundException: Unable to find an entry point named 'thunderscopeAdcCalibrationSet' in shared library 'tslitex'.
   at TS.NET.Driver.Libtslitex.Interop.SetAdcCalibration(IntPtr ts, tsAdcCalibration_t& cal)
   at TS.NET.Driver.Libtslitex.Thunderscope.SetAdcCalibration(ThunderscopeAdcCalibration adcCal) in /home/thunderscope/ThunderScope/Software/TS.NET/source/TS.NET/Drivers/Libtslitex/Thunderscope.cs:line 284
   at TS.NET.Driver.Libtslitex.Thunderscope.Open(UInt32 devIndex, ThunderscopeHardwareConfig initialHardwareConfiguration) in /home/thunderscope/ThunderScope/Software/TS.NET/source/TS.NET/Drivers/Libtslitex/Thunderscope.cs:line 46
   at Program.Start(Int32 deviceIndex, String configurationFilePath, Int32 seconds, Boolean membench) in /home/thunderscope/ThunderScope/Software/TS.NET/source/TS.NET.Engine/Program.cs:line 169
   at System.CommandLine.Handler.<>c__DisplayClass5_0`4.<SetHandler>b__0(InvocationContext context)
   at System.CommandLine.Invocation.AnonymousCommandHandler.Invoke(InvocationContext context)
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext context)
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass17_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass12_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__18_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__5_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass8_0.<<UseExceptionHandler>b__0>d.MoveNext()
`
With the current main-branch of libtslitex, the io-test works as it did before, at least with the corresponding test-scripts.
The latest dev-updates does not work correctly (I built and flashed the latest gateware AFAIK), the I2C-Test doesn't execute and TS.NET cannot start. This is the output of the dev-updates io-test:

`Opening Device /dev/thunderscope0
[> Device Information:
------------------------
FPGA Identifier:  LitePCIe SoC on ThunderScope A200T (6508cf6e) 2025-06-16 16:36:51.
HW Rev Beta
FPGA DNA:         0x003465012ef9e85c
FPGA Temperature: 44.2 �C
FPGA VCC-INT:     1.01 V
FPGA VCC-AUX:     1.79 V
FPGA VCC-BRAM:    1.01 V
[> Scratch register test:
-------------------------
Write 0x12345678 to Scratch register:
Read: 0x12345678
Write 0xdeadbeef to Scratch register:
Read: 0xdeadbeef
Enabling frontend LDO:
Enabling ADC LDO:
Disabling PLL EN & waiting 500ms:
Enabling PLL EN:
Set READY LED
Set ERROR LED
Set ACTIVE LED

Scan I2C Bus 0:

0x00 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
0x10 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
0x20 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F
0x30 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
0x40 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F
0x50 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F
0x60 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F
0x70 70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F
Press ENTER to continue...`

How can I know which HW/SW/GW states fit together? I had a setup running back in February, but it somehow stopped working...

BR
Sebastian

AleksaBjelogrlic, 2025-06-17:
Oh yes there were some calibration changes @NateMeyer did that may not have been rolled in on @macaba's side yet

NateMeyer, 2025-06-17:
The `dev_updates` branch of libtslitex tracks with the `dev_build` branch of the thunderscope_litex gateware.  These will get merged into main when the gamma checkout is completed.

The main branch of TS.NET is tracking the API changes the above dev branches.

@Vrail99 do you have any more info on the kernel 6.11 issue?

Looks like you are on an older commit of the gateware main branch.  Can you try running with at least commit 37c994d of the gateware and the latest `dev` branch driver?

TODO: I think the driver API should be stable enough to roll those into main for a 1.0 release

Vrail99, 2025-06-17:
Yes, I just tried installing the ts_litex driver from the main branch: 

`
Already up to date.
make: Nothing to be done for 'all'.
[sudo] password for thunderscope: 
Kernel may have changed, try to rebuild module
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/main.c:300:6: warning: no previous prototype for ‘litepcie_stop_dma’ [-Wmissing-prototypes]
  300 | void litepcie_stop_dma(struct litepcie_device *s)
      |      ^~~~~~~~~~~~~~~~~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/main.c:954:6: warning: no previous prototype for ‘sfind’ [-Wmissing-prototypes]
  954 | void sfind(char *string, char *format, ...)
      |      ^~~~~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/main.c:969:5: warning: no previous prototype for ‘compare_revisions’ [-Wmissing-prototypes]
  969 | int compare_revisions(struct revision d1, struct revision d2)
      |     ^~~~~~~~~~~~~~~~~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c: In function ‘liteuart_start_tx’:
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:150:45: error: ‘struct uart_state’ has no member named ‘xmit’
  150 |         struct circ_buf *xmit = &port->state->xmit;
      |                                             ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:157:21: error: implicit declaration of function ‘uart_circ_empty’; did you mean ‘uart_lsr_tx_empty’? [-Werror=implicit-function-declaration]
  157 |         } else if (!uart_circ_empty(xmit)) {
      |                     ^~~~~~~~~~~~~~~
      |                     uart_lsr_tx_empty
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:158:28: error: invalid use of undefined type ‘struct circ_buf’
  158 |                 while (xmit->head != xmit->tail) {
      |                            ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:158:42: error: invalid use of undefined type ‘struct circ_buf’
  158 |                 while (xmit->head != xmit->tail) {
      |                                          ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:159:34: error: invalid use of undefined type ‘struct circ_buf’
  159 |                         ch = xmit->buf[xmit->tail];
      |                                  ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:159:44: error: invalid use of undefined type ‘struct circ_buf’
  159 |                         ch = xmit->buf[xmit->tail];
      |                                            ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:160:29: error: invalid use of undefined type ‘struct circ_buf’
  160 |                         xmit->tail = (xmit->tail + 1) & (UART_XMIT_SIZE - 1);
      |                             ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:160:43: error: invalid use of undefined type ‘struct circ_buf’
  160 |                         xmit->tail = (xmit->tail + 1) & (UART_XMIT_SIZE - 1);
      |                                           ^~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:166:13: error: implicit declaration of function ‘uart_circ_chars_pending’ [-Werror=implicit-function-declaration]
  166 |         if (uart_circ_chars_pending(xmit) < WAKEUP_CHARS)
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c: At top level:
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:359:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Werror=incompatible-pointer-types]
  359 |         .remove = liteuart_remove,
      |                   ^~~~~~~~~~~~~~~
/home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.c:359:19: note: (near initialization for ‘liteuart_platform_driver.<anonymous>.remove’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:244: /home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel/liteuart.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.11.0-26-generic/Makefile:1927: /home/thunderscope/ThunderScope/Software/ts_litex_driver_linux/kernel] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make: *** [Makefile:14: litepcie.ko] Error 2
`

Seems like some kernel functions were changed.

BR,
Sebastian