EEVengers/ThunderScope #370: Calibrate Rev 5.1 Units

AleksaBjelogrlic, 2025-08-29:
As I calibrate these units, I will be documenting the theory and procedure on the [docs page](https://docs.eevengers.com/ThunderScope/latest/Theory-of-Operation/Calibration.html) and putting the results here.

AleksaBjelogrlic, 2025-09-22:
Use Mark's calibration program to calibrate all three (once the other two are assembled) Rev 5.1 units

AleksaBjelogrlic, 2025-10-20:
The one 5.1 unit that I built (PCIe unit, TS0023) calibrates successfully with Mark's program

s070388s, 2025-11-14:
How should I calibrate Passband Gains and DC Offsets? I've tried many methods but haven't figured it out. Could you please tell me the way?

AleksaBjelogrlic, 2025-11-14:
You can run the calibration through TS.NET.Testbench, which you can build using the script here: https://github.com/EEVengers/TS.NET/tree/main/build-scripts

s070388s, 2026-10-14:
Thanks, I've been busy with other things and just found some time. I'll give it a try later.